ZinggJM/GxEPD2 #1: GxEPD2_WiFi_Example ESP8266 not displaying downloaded bitmap

noelportugal, 2018-07-15:
### Expected Behavior
Display downloaded bitmap on EPD module. 

### Actual Behavior
It downloads the BMP, but it doesn't display it. I can verify that display is connected correctly because I can print text to it (I added display.print("Downloading file "); display.println(file_name); before downloading the actual file.)

### E-Paper module used
Waveshare 200x200, 1.54inch E-Ink display module, three-color. 
[](https://www.waveshare.com/product/1.54inch-e-paper-module-b.htm)

### Arduino / Processor module used
NodeMCU 1.0 (ESP-12E Module) 4M (1M SPIFFS)
CS=D8, DC=D3, RST=D4, BUSY=D2, DIN=D7, GND=GND, 3.3v=3.3v

### Steps to reproduce the behavior or reduced but complete code example
I'm using a modified version of the Libary GxEPD2_WiFi_Example. I am just downloading one bitmap image from the github repo. Here is the sketch I'm using:

[](https://gist.github.com/noelportugal/e32f5030ee15e1d1eb0c38972ff8efd5)

**Serial output:**
downloading file "logo200x200.bmp"
connecting to raw.githubusercontent.com
certificate matches
requesting URL: https://raw.githubusercontent.com/ZinggJM/GxEPD2/master/extras/bitmaps/logo200x200.bmp
request sent
HTTP/1.1 200 OK
headers received
File size: 5662
Image Offset: 62
Header size: 40
Bit Depth: 1
Image size: 200x200
downloaded in 32894 ms
bytes read 5662

At the end, the EPD refreshes with white and no image is displayed.

BTW thanks for a great library...I have been using GxEPD, but I have been really looking forward to the ability to display remote images.

ZinggJM, 2018-07-16:
The 1.54" 200x200 3-color e-paper display has no partial update capability; the controller has no capability to address parts of the internal RAM.
The bitmap file drawing examples use the partial update capability to write to the controller RAM.
For the example GxEPD2_SD_Example I put a comment that it can't be used with GxEPD2_154c, but I forgot to add this comment to the other examples.
I will later add support for file bitmap drawing through buffered drawing, except for AVR Arduino.

noelportugal, 2018-07-16:
Thanks for your prompt reply! Fortunately, this is not the EDP I will plan to use for my project. I'm just waiting for another (B&W) EDP on the mail... I will close this now...Thanks!

closed 2018-07-16 by noelportugal