ZinggJM/GxEPD2 #2: No images displayed using wifi example

vincentjanv, 2018-10-10:
## Please post only issues that are real issues with this library!
- please use the Arduino Forum Displays for questions or enhancement suggestions

### Expected Behavior
Images being displayed on screen
### Actual Behavior
Screen refreshes but ends up white, never is an image (or part of an image) shown
### E-Paper module used
Waveshare 4.2inch 3 color (yellow)
### Arduino / Processor module used
Waveshare wifi driver board (ESP8266)
### Steps to reproduce the behavior or reduced but complete code example

1. Enter WiFi information
2. select correct board (by removing comment)
`GxEPD2_3C<GxEPD2_420c, GxEPD2_420c::HEIGHT / 2> display(GxEPD2_420c(/*CS=15*/ SS, /*DC=4*/ 4, /*RST=5*/ 5, /*BUSY=16*/ 16));
3. Compile and upload
4.Console output shows correct downloading of images
5. Screen starts refreshing for each image, but always ends up completely white (before loading new image)

Using the source code provided by WaveShare (html server to easily upload image), the screen refreshes fine.
`

vincentjanv, 2018-10-10:
Other examples are showing fine.

This is the console output:

`18:02:07.793 -> downloading file "first200x200.bmp"
18:02:07.793 -> connecting to raw.githubusercontent.com
certificate matches
18:02:08.211 -> requesting URL: https://raw.githubusercontent.com/ZinggJM/GxEPD2/master/extras/bitmaps/first200x200.bmp
18:02:08.211 -> request sent
HTTP/1.1 200 OK

18:02:08.423 -> headers received
18:02:08.423 -> File size: 5662
18:02:08.423 -> Image Offset: 62
18:02:08.423 -> Header size: 40
18:02:08.423 -> Bit Depth: 1
18:02:08.423 -> Image size: 200x200
Busy Timeout!
18:02:28.712 -> _Update_Part : 20000782
downloaded in 22564 ms
18:02:30.337 -> bytes read 5662
_Update_Full : 4602283
`

ZinggJM, 2018-10-10:
> Other examples are showing fine.

Then I don't understand why you get 

> Busy Timeout!

I would need more diagnostic output to analyze.

vincentjanv, 2018-10-11:
Hi,
thanks for replying. Could you point me to a place where I can get a more diagnostic output for you? Many thanks in advance!

ZinggJM, 2018-10-11:
> I would need more diagnostic output to analyze.

I meant more length for a start, to see if "Busy Timeout!" occurs more often.
GxEPD2_WiFi_Example.ino  shows many bitmaps, so you get long diagnostic output in Serial Monitor.
And just for completeness, post the version of GxEPD2 you use.
You could also compare diagnostic output of GxEPD2_Example with the one from GxEPD2_WiFi_Example yourself, **with the exact same configuration**, same board, same board selected for build.

ZinggJM, 2018-11-03:
No more input. Closed.

closed 2018-11-03 by ZinggJM