ZinggJM/GxEPD2 #3: Waveshare 9.7 1200x825 display

Bluebrain2000, 2018-11-18:
Is it possible to add support for the Waveshare 9.7 1200x825 display?
https://www.waveshare.com/product/modules/oleds-lcds/e-paper/9.7inch-e-Paper.htm

It looks like a great display with 16 shades of greyscale support and a very fast refresh rate of less than a second.

ZinggJM, 2018-11-18:
This display is indeed very interesting. The panel is from Dalian Good Display. The panel has a parallel interface, so neither GxEPD2 nor GxEPD would be directly suitable to support it. GxEPD_HD would be the suitable library to support it, either with the DESTM32-T evaluation board from Good Display or the adapter board from Waveshare.

The price for this e-paper display will not make it used by many Arduino or hobby users. So  although I am interested in this display and in supporting it, I would need a sponsor or donator to get me such a display. Neither Waveshare nor Good Display showed interest in sponsoring me, not even for the new small SPI e-paper displays; free evaluation samples are only available for volume customers.

ZinggJM, 2019-01-16:
No sponsor or donator, so no support in the library, as explained in README.MD.

closed 2019-01-16 by ZinggJM