ZinggJM/GxEPD2 #4: setPartialWindow() limited in x?

crteensy, 2019-01-04:
I'm not sure if this is library behavior or a limitation of the display as such.

After drawing a frame (in my case: for a plot) and then setting the partial update window to be just the inside of that frame, I expect the frame to be left intact so that I can just update the plot area.

However, after setting partial update mode, the frame is partially gone. The partial update window seems to be larger, but only in x direction. The y direction seems to be alright. Furthermore, the horizontal line drawn in the example below seems to be about 4 pixels wide (but not consistently).

Display: Waveshare 4.2" black/white

Board: nodeMCU with esp-12E module (ESP8266)

Arduino Sketch:
```
#define ENABLE_GxEPD2_GFX 0

#include <GxEPD2_BW.h>

// nodemcu with ESP8266
#define DISPLAY_CS_PIN     0
#define DISPLAY_DC_PIN     4
#define DISPLAY_RST_PIN    5
#define DISPLAY_BUSY_PIN  16
typedef GxEPD2_BW<GxEPD2_420, GxEPD2_420::HEIGHT> TDisplay;
TDisplay display(GxEPD2_420(DISPLAY_CS_PIN, DISPLAY_DC_PIN, DISPLAY_RST_PIN, DISPLAY_BUSY_PIN));

void setup() {
  Serial.begin(115200);

  display.init(115200);
  display.setRotation(0);

  display.setFullWindow();
  display.firstPage();
  do
  {
    display.fillScreen(GxEPD_WHITE);
  }
  while (display.nextPage());

  do
  {
    display.drawRect(3, 3, 100, 75, GxEPD_BLACK);
  }
  while (display.nextPage());

  display.setPartialWindow(4, 4, 98, 75);
  display.firstPage();
  do
  {
    display.drawFastHLine(0, 39, 400, GxEPD_BLACK);
    display.drawFastVLine(52, 0, 300, GxEPD_BLACK);
  }
  while (display.nextPage());  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
[image]

ZinggJM, 2019-01-08:
Your issue contains 2 issues, and both are real issues:

1: there is a comment missing on the setPartialWindow() method:

`
  // setPartialWindow; x and w should be multiple of 8, else window is increased as needed,
`
`
  // this is an addressing limitation of the e-paper controllers
`
`
    void setPartialWindow(uint16_t x, uint16_t y, uint16_t w, uint16_t h)
`

2: there is a clipping to the partial window boundary missing in drawPixel(), which causes strange effects when partial window is set, but drawing functions draw outside the limit. The exact behavior of the effect is unclear.

Both issues will be fixed with the next release.

Thank you for reporting these issues.

ZinggJM, 2019-01-16:
Fixed in Version 1.1.2

closed 2019-01-16 by ZinggJM