ZinggJM/GxEPD2 #5: ESP8266HTTPUpdateServer Stops working after issuing display.powerOff()

sonic2wb, 2019-01-13:
## Please post only issues that are real issues with this library!
- please use the Arduino Forum Displays for questions or enhancement suggestions

### Expected Behavior
Display to enter low power state
### Actual Behavior
No error from the display but other ESP8266 functions quit working
### E-Paper module used
Waveshare 2.7 in Hat
### Arduino / Processor module used
ESP8266 esp-12f

### Steps to reproduce the behavior or reduced but complete code example
code:  https://pastebin.com/HScxbFNa

As long as i dont call display.powerOff(); everything works correctly. but once its called trying to upload new data via http://IP.of.esp/update  fails with connection reset.

sonic2wb, 2019-01-13:
[image]

ZinggJM, 2019-01-13:
Try different connection pins for the e-paper. Especially SS for CS may be the cause, as GPIO15 is used also for boot mode selection on reset, I think for OTA boot mode. It is unfortunate that this collides with the standard SS definition, which is also the standard HW SPI CS pin (the HW SPI could automatically produce CS on that pin when programmed for that, afaik).

sonic2wb, 2019-01-13:
OK, I have already moved CS from 15 to 02 as it was hanging the normal boot. 
Just to add it fails during the upload of the new .bin file. its like powerOff is tieing up the processor. 
I will try moving CS to 01 or 03 but will lose the serial console. Its not a deal breaker by no means just an odd thing.

ZinggJM, 2019-01-16:
> is there way to disable the debug for the display?

`    void init(uint32_t serial_diag_bitrate = 0) // = 0 : disabled
`
> BUT if i disable Serial.begin. everything works as normal. 

Strange, as display.init(115200); enables serial

In my opinion, this is **not** an issue with the library, but maybe a wiring issue, or maybe Serial can't be used with OTA?

closed 2019-01-16 by ZinggJM