ZinggJM/GxEPD2 #7: Issue with WiFi example (invalid conversion of output_row_colow_bufer)

TheFitzZZ, 2019-02-02:
### Expected Behavior
Being able to compile the sketch and draw the example picture.

### Actual Behavior
Upon compiling, I get the error
`invalid conversion from 'uint8_t* {aka unsigned char*}' to 'int16_t {aka short int}' [-fpermissive]`
from both lines
`display.writeImage(output_row_mono_buffer, output_row_color_buffer, x, yrow, w, 1);`
I have tested which variable is the culprit, and it seems to be output_row_color_buffer, as it compiles fine (but not obviously not displaying anything) when I replace it with a int16_t variable.
I tried to understand how it is put together, but it's beyond my current capability of comprehending the code.

### E-Paper module used
Waveshare 7.5 with e-Paper hat. Confirmed working with the basic example.

### Arduino / Processor module used
NodeMcu V3 / 8266 - in the Ardruino Gui I choose NodeMCU 1.0 ESP-12E Mod

### Steps to reproduce the behavior or reduced but complete code example
Just added my line to address the display:
`#if defined (ESP8266)
GxEPD2_3C<GxEPD2_750c, GxEPD2_750c::HEIGHT / 4> display(GxEPD2_750c(D8,  D3, D4, D2));
#endif`

Thank you for your time,
//Chris

ZinggJM, 2019-02-03:
@TheFitzZZ 

Thank you for reporting this error.

This error was introduced with the last release, when fixing an error reported by some compiler complaining about ambiguous overloaded methods for writeImage(...) and drawImage().
I had removed the default parameter values for the 3-color methods, and did not check with all examples. 
I will fix this error in the next release.

In the meantime you can add the 3 missing parameter values explicitly on lines 548 and 1025 in the GxEPD2_WiFi_Example:
`          display.writeImage(output_row_mono_buffer, output_row_color_buffer, x, yrow, w, 1, false, false, false);
`
The same error occurs with GxEPD2_SD_Example, GxEPD2_SD_AVR_Example, GxEPD2_Spiffs_Example.

TheFitzZZ, 2019-02-03:
Perfect - works! Thank you very much for the swift response!

closed 2019-02-03 by TheFitzZZ