ZinggJM/GxEPD2 #8: drawPaged() calls callback twice

Lithimlin, 2019-02-06:
Whenever I call `drawPaged(myCallback, 0);`, the function `myCallback(...)` is called twice.

### Expected Behavior
`drawPaged(myCallback, 0);` calls the function `myCallback(...)` once
### Actual Behavior
`drawPaged(myCallback, 0);` calls the function `myCallback(...)` twice
### E-Paper module used
2.9 inch e-Paper Module
### Arduino / Processor module used
Esp32
### Steps to reproduce the behavior or reduced but complete code example
```
void showQuestionnaire(Questionnaire q){
  display.setFullWindow();
  display.setRotation(1);
  u8g2Fonts.setFontMode(1);
  u8g2Fonts.setFontDirection(0);
  u8g2Fonts.setForegroundColor(GxEPD_BLACK);
  u8g2Fonts.setBackgroundColor(GxEPD_WHITE);

  u8g2Fonts.setFont(questionFont);
  Serial.println("drawing Question");
  display.drawPaged(showQuestionCallback, &q);
  Serial.println("Question drawn");

  //code that calculates the variable offset

  display.setPartialWindow(0, display.height()-offset, display.width(), offset);
  u8g2Fonts.setFont(answerFont);
  Serial.println("drawing Answers");
  display.drawPaged(showAnswersCallback, &q);
  Serial.println("finished drawing");
}

void showQuestionCallback(const void* hQuestion) {
  Serial.println("draw question callback");
  display.fillScreen(GxEPD_WHITE);
  uint8_t y = u8g2Fonts.getFontAscent() + 1;
  u8g2Fonts.setCursor(0, y);
  //some code formatting and printing the question...
}

void showAnswersCallback(const void* hAnswers) {
  Serial.println("draw answers callback");
  display.setRotation(0);
  //some code formatting and printing the answers...
}
```

### Output on the serial monitor
```
drawing Question
draw question callback
_PowerOn : 71523
_Update_Full : 1556573
draw question callback
_Update_Part : 406814
Question drawn
drawing Answers
draw answers callback
_Update_Part : 406821
draw answers callback
_Update_Part : 406821
finished drawing
```

ZinggJM, 2019-02-06:
You didn't explain why this is an issue for you. Speed concerns?

This behaviour occurs with displays that provide "fast partial update", pure differential update.
For this to work, the "previous" buffer and the "current" buffer in the controller need to be set equal after an update, the controllers does not copy buffers by itself (most do switch the buffers), else the next update would produce garbage. For real paged drawing, this requires a second round of calls of the callbacks.
This could be optimised for the special case of full buffered drawing (with one page only), but is not implemented for callback use, only for firstPage..nextPage use (who was going to use callbacks anyway).
The additional refresh (_Update_Part) with no difference is not really needed, but I was not sure at the time of implementation. (Initially I had only drawImage() in the controller classes, no writeImage()).

I keep the issue open, as I intend to remove the additional refresh, and test this.

Lithimlin, 2019-02-06:
In my `showQuestionCallback()`, I count the lines I need to print the question to adjust the size of the answer area. I simply incremented it in the callback, so I always ended up with twice the amount of lines. I'm just set it to 0 at the start of the callback now.

ZinggJM, 2019-02-10:
Removed the additional refresh, and optimized for full buffered.

closed 2019-02-10 by ZinggJM