ZinggJM/GxEPD2 #11: Display Issue with Line that comes without intention....

njordan77, 2019-03-04:
i have a 7,5inch display that is updated via ESP8266 + GxEPD2. As far as I remember initially the display worked fine but after several months i have a strange line in the display that does not go away anymore....
I update the display every 20mins, so this should not be too often i guess....

Any idea what could be wrong or if a special command is required to kind of reset the display or force a cleaning process......or is the display broken?

[image]

ZinggJM, 2019-03-05:
Does your e-paper still show normal update behaviour with ~4s flashing?
Do you use partial update? partial update only?

Do you also see artifacts when running GxEPD2_Example? It looks like one line driver stays constantly active.

What happens if you completely remove power for a minute and then restart.

My 7.5" b/w is in constant use for over a year now, with update every 3 minutes. I use only full update.
It still looks perfect.

njordan77, 2019-03-05:
Hello, thanks for the quick reply!
No, i do not use partial updates (never did) - always full updates. Not sure what you mean by the~4s flashing. The display flashes like always several times until the final state arrives.....like you wrote ~4s.
Do not know the tech details, but if a line driver is broken - wouldn't in full update a complete line from 0 to max horizontal happen?

Removal of power and reconnect does not change anything. But good to know that "every 15mins" full updates are not "the killer".

ZinggJM, 2019-03-06:
A line driver stuck a some level would mean that pixels in that line are affected by pixel values of other lines, but only in one direction, e.g. to white like in your picture. But I am not a HW expert.

Your display really seems broken, and this is not an issue of the library.

closed 2019-03-06 by ZinggJM