ZinggJM/GxEPD2 #13: WEMOS D1 MAPPING CHANGE

lahoura, 2019-03-21:
## Please post only issues that are real issues with this library!
- please use the Arduino Forum Displays for questions or enhancement suggestions

### Expected Behavior

### Actual Behavior

### E-Paper module used

### Arduino / Processor module used

### Steps to reproduce the behavior or reduced but complete code example

lahoura, 2019-03-21:
After trying to use your work with WEMOS D1 if you connect :

// mapping suggestion from Waveshare SPI e-Paper to Wemos D1 mini
// BUSY -> D2, RST -> D4, DC -> D3, CS -> D8, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V
it doesn't work and you can't upload sketch with arduino IDE

see :https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup/

if you use :
// mapping suggestion from Waveshare SPI e-Paper to Wemos D1 mini
// BUSY -> D2, RST -> D6, DC -> D0, CS -> D4, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V
line 64: ( for my 1.54inch e-Paper waveshare module 2color )
GxEPD2_3C<GxEPD2_154c, GxEPD2_154c::HEIGHT> display(GxEPD2_154c(/*CS=D4*/ 2, /*DC=D0*/ 16, /*RST=D6*/ 12, /*BUSY=D2*/ 4));
it works great !
Best..

ZinggJM, 2019-03-22:
**Ok, I will keep this issue open forever, to let you get famous.**

See also line 25 of GxEPD2_Example:

`// NOTE for ESP8266: using SS (GPIO15) for CS may cause boot mode problems, use different pin in case
`

TondeuseDolly, 2019-05-19:
Hi there !

It works also for me on Wemos D1 Mini + 4.2 inches b/w ePaper (GxGDEW042T2/GxGDEW042T2.h)
Actually it's the CS pin that blocked the upload and program !

The only problem of your solution is that PIN D0 is now used so we cannot use it to wake up the ESP8266 when in sleep mode. 
So I changed a bit your proposal :
```
  // BUSY -> D2, RST -> D6, DC -> D1, CS -> D4, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V
  
  GxIO_Class  io      (SPI, /*CS*/ D4, /*DC*/ D1, /*RST*/ D6);
  GxEPD_Class display (io, /*RST*/ D6, /*BUSY*/ D2);
```

Then ESP.deepSleep is working correctly !

Regards