ZinggJM/GxEPD2 #15: On ESP32: Can only use VSPI - how to access HSPI?

jtec, 2019-05-02:
Hi, 

- The library uses the global "SPI" object, which in turn uses VSPI on the ESP32.
- I manually added 

SPIClass SPI2(HSPI);

in line 21 of GxEPD_EDP.cpp and replaced all occurences of "SPI." by "SPI2."

- In the GxEPD library, the SPI interface could be selected, is there a way to do so in GxEPD2 without modifying the library code?

- Processor: ESP32
- Display 1.54" 200x200 b/w

cheers
Jan

ZinggJM, 2019-05-03:
You do not give a reason why you need to use HSPI on ESP32.
The design goal of GxEPD2 is to keep it simple. 
With ESP32 you can map SPI to alternate IO pins, see example GxEPD2_WS_ESP32_Driver.
No, you can't select a different SPI HW module in GxEPD2 without modification to the library.
The Library GxEPD is more flexible by design.

closed 2019-05-03 by ZinggJM

jtec, 2019-05-03:
Hi, 

thanks for your rapid answer.

best regards, 
Jan