ZinggJM/GxEPD2 #20: Compile error when using PlatformIO

sidoh, 2019-05-18:
### Expected Behavior

Compiles successfully

### Actual Behavior

It doesn't compile:

```
from lib/Display/BitmapRegion.cpp:2:
.piolibdeps/GxEPD2_ID5869/src/GxEPD2_GFX.h:20:26: error: expected ')' before '&' token
GxEPD2_GFX(GxEPD2_EPD& _epd2, int16_t w, int16_t h) : Adafruit_GFX(w, h), epd2(_epd2) {};
^
In file included from lib/Display/Region.h:5:0,
from lib/Display/BitmapRegion.h:1,
from lib/Display/BitmapRegion.cpp:2:
.piolibdeps/GxEPD2_ID5869/src/GxEPD2_GFX.h:72:5: error: 'GxEPD2_EPD' does not name a type
GxEPD2_EPD& epd2;
^
In file included from lib/Display/Region.h:5:0,
from lib/Display/Region.cpp:1:
.piolibdeps/GxEPD2_ID5869/src/GxEPD2_GFX.h:20:26: error: expected ')' before '&' token
GxEPD2_GFX(GxEPD2_EPD& _epd2, int16_t w, int16_t h) : Adafruit_GFX(w, h), epd2(_epd2) {};
```

I believe all that's needed is an include in `GxEPD_GFX.h`:

https://github.com/ZinggJM/GxEPD2/compare/master...sidoh:partial_display#diff-f66df49d4b0357a22bea5c28e88f478c

```c++
...
#ifndef _GxEPD2_GFX_H_
#define _GxEPD2_GFX_H_

// Add include here:
#include "GxEPD2_EPD.h"

#include <Adafruit_GFX.h>
...
```

### E-Paper module used

GDEW042T2

### Arduino / Processor module used

ESP32

### Steps to reproduce the behavior or reduced but complete code example

Compiling this project:

https://github.com/sidoh/epaper_templates/tree/http_refactor

Using published GxEPD2 library instead of my github fork.

Compiled using `pio run -e esp32`.

ZinggJM, 2019-05-18:
Confirm, this #include is missing, as it should be enough to #include GxEPD2_GFX.h to use the base class, e.g. in any separate .cpp file.
Ok, fine if GxEPD2 can be used with PlatformIO, but I do not really care. I claim it can be used with Arduino IDE.
Thanks for reporting.

sidoh, 2019-05-18:
re: PlatformIO - just being explicit about my environment in case it's helpful.

And for what it's worth, I've never encountered a case where something compiles with the Arduino IDE, but not using PlatformIO.  PIO is generally easier, actually, because it automatically resolves upstream dependencies.

ZinggJM, 2019-05-31:
Added in Version 1.1.8

closed 2019-05-31 by ZinggJM