ZinggJM/GxEPD2 #21: displayWindow method added in 1.1.7 is not exposed in base class

sidoh, 2019-05-18:
### Expected Behavior

`displayWindow` method added in 1.1.7 should be exposed in `GxEPD2_GFX` base class.

Think we just need this:

https://github.com/ZinggJM/GxEPD2/compare/master...sidoh:partial_display?expand=1#diff-f66df49d4b0357a22bea5c28e88f478cR35

```c++
    virtual void displayWindow(uint16_t x, uint16_t y, uint16_t w, uint16_t h) = 0;
```

in `GxEPD2_GFX.h`.

### Actual Behavior

Method is not in the base class, so code using base class can't reference this method.

### E-Paper module used

4.2" b/w waveshare

### Arduino / Processor module used

ESP32

### Steps to reproduce the behavior or reduced but complete code example

https://github.com/sidoh/epaper_templates/tree/http_refactor

ZinggJM, 2019-05-18:
Confirm, forgot to add this method to the GxEPD2_GFX base class. Thanks for reporting.

ZinggJM, 2019-05-31:
Added in Version 1.1.8

closed 2019-05-31 by ZinggJM