ZinggJM/GxEPD2 #22: GDEW075Z09 7.5" b/w/r and RED color

ok1cdj, 2019-05-28:
With this display not working drawing in RED color. Black and white is ok, but red is
gray. By datasheet this display accept 7 colors . I think there is some problem with color definition.
  
### E-Paper module used
GDEW075Z09 7.5" b/w/r 

### Arduino / Processor module used
ESP8266

ZinggJM, 2019-05-28:
Do you really have a GDEW075Z09, the 3-color version 7.5" e-paper display?
Did you test with the GxEPD2_Example?
Did you select the correct constructor for the 3-color version?
`GxEPD2_3C<GxEPD2_750c, GxEPD2_750c::HEIGHT / 4> display(GxEPD2_750c(/*CS=D8*/ SS, /*DC=D3*/ 0, /*RST=D4*/ 2, /*BUSY=D2*/ 4));
`
The display does not accept 7 colors, only one color and 4 levels of grey.
GxEPD2 supports grey levels only for bitmaps containing native format with method drawNative().
All graphics drawing only takes color values GxEPD_BLACK, GxEPD_WHITE, GxEPD_RED.

ok1cdj, 2019-05-28:
yes, constructor is OK. yes 3 color and 4 levels of grey.
example also not display red color, red color is represented as light grey

ZinggJM, 2019-05-28:
The RESE switch or jumper on your connection board may be set to the wrong position; needs to be on 0.47 ohms. What connection board do you use?
Where did you buy your panel? Do you see red ink in the corners on the connector side?
You should see long flashing on update ~30 seconds.
I just checked again with my 7.5" 3-color red panel; I see red text and red bitmap parts.

ok1cdj, 2019-05-28:
yes, 0.47 ohms. I have one comercial break out and my one. Both same results. I have red ink on corners. No 30 sec update, update is much faster. I have also 2.3 ink with red color and this working fine.
Display is from same source. 
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.37352e8dXAxw81&id=564915001652&_u=s2u2lmbk399e

ZinggJM, 2019-05-28:
I think there is something wrong with your display. 
If "much faster" is ~4.3s then you selected the constructor for b/w. Or your display has a wrong wavetable in OTP.
Take a look at diagnostic output on Serial Monitor, and post a part that shows update times.

ZinggJM, 2019-05-29:
This is **not** an issue of this library. Closed.

closed 2019-05-29 by ZinggJM