ZinggJM/GxEPD2 #24: Display not showing text or respecting rotation

rlightner, 2019-06-05:
#### Version number of the library used, should be the current one
1.1.8

#### Expected Behavior
Image would be drawn with text below

#### Actual Behavior
Image is drawn upside down (not respecting rotation), text is not displayed

#### E-Paper module used
TTGO 2.7in

#### Arduino / Processor module used
ESP32

#### Steps to reproduce the behavior or reduced but complete code example

    String msg = getJoinWifiMessage();
    display.init(115200);
    display.clearScreen(); // white
    display.setRotation(2); //font orientation
    display.setFullWindow();
    display.fillScreen(GxEPD_WHITE);
    
    display.drawImage(logo_w_text, 0, 0, 100, 136);

    display.setFont(&FreeMonoBold9pt7b);
    display.setTextColor(GxEPD_BLACK);
    int16_t tbx, tby;
    uint16_t tbw, tbh;
    display.getTextBounds(msg, 0, 0, &tbx, &tby, &tbw, &tbh);
    // center bounding box by transposition of origin:
    uint16_t x = ((display.width() - tbw) / 2) - tbx;
    uint16_t y = ((display.height() - tbh) / 2) - tby;
    // full window mode is the initial mode, set it anyway
    display.setCursor(x, y);
    display.print(msg);
    
    display.refresh(true); // full update

rlightner, 2019-06-05:
if I comment out the drawImage (I've also tired writeImage) the text will render correctly.

rlightner, 2019-06-08:
Still having issues with rotation not being respected. If I run the GxEPD2_SD_Example an set the rotation, text prints correctly, but the images are drawn "upside down"

ZinggJM, 2019-06-08:
… and you still believe this is an error of this library.

Please read the README.MD and try to understand the difference between direct drawing of bitmaps to the screen (controller buffer) and graphics drawing to the graphics buffer. Rotation is a feature of graphics drawing (Adafruit_GFX). And Adafruit_GFX has methods to draw bitmaps also. GxEPD2_Example shows the use of both kinds of bitmap drawing.
 GxEPD2_SD_Example uses direct drawing to the controller line per line. But it also has the code for buffered drawing because of the 1.54" 3-color 200x200 display that has no window addressing capability.

closed 2019-06-08 by ZinggJM